space-concordia-robotics/robotics-rover #2: Install OS on Rover

psyomn, 2015-05-11:
Soon it shall be linux. And tomorrow all will be linux.

Alright, on a more serious note, go ahead and install whatever OS you see fit, as long as it's not a huge resource hog. Make sure that it has proper drivers for network, since we'll want to connect to it soon. Make sure it has python2.7, which is what we're using. Here's a list of todos:

[ ] Can connect to internets
[ ] Can has python2.7
[ ] Can has git (we might need this)

If you have any comments, or find oddities, follow up with a discussion here.

Doomfest, 2015-05-11:
why do you need git on the rover? also theres no internet on the rover

psyomn, 2015-05-11:
This is strictly for dependency+development. We'll be pulling in the libs from github, and running the code like that. We won't want this during production/the competition however.

Doomfest, 2015-05-11:
Okay that makes sense but there is still the problem of no internet access to that router....

psyomn, 2015-05-11:
Oh right. School network. Yay. Hmm. I'll think of something twisted when the time comes.

Doomfest, 2015-05-11:
I'll have to check how the routers are connected together. We might be able to just put an ethernet cable.

msnidal, 2015-05-22:
Note: Internet on rover through Concordia Wi-fi to my phone and USB tethering

closed 2015-05-22 by msnidal

psyomn, 2015-05-23:
^ Arielle wrote wiki about this https://github.com/space-concordia-robotics/robotics-rover/wiki

Also, we are probably installing some other OS. I'll be happy to help whenever you're there Mark!

Doomfest, 2015-05-23:
What is wrong with the current OS?

psyomn, 2015-05-23:
I ... broke it :(.