space-concordia-robotics/robotics-rover #4: Video Streaming

arevans3, 2015-05-22:
board needs to take video and stream it somehow. vlc?

Doomfest, 2015-05-22:
I dont know how you guy are going to use vlc to stream it. I thought we were going to use the networking protocol we were building.

psyomn, 2015-05-24:
So for this, it might make sense to reuse some parts of vlc to capture streams perhaps (if it doesn't make sense, maybe something else exists we can use, so some research here might be needed). There will be another component needed where the compression happens.

Doomfest, 2015-05-25:
Theres a compression library for python, but I don't how we will use it with vlc...

psyomn, 2015-05-25:
I'll probably be at lab tomorrow if you want to meet up!

arevans3, 2015-06-02:
can stream video with vlc using wizard. closing issue

closed 2015-06-02 by arevans3