space-concordia-robotics/robotics-rover #8: Arch-Linux

arevans3, 2015-05-24:
make sure usb tethering works so we can have internet right away. check wiki.

psyomn, 2015-05-24:
@msnidal (make sure we get the iso which is not netinstall (if that still exists)). Ping me whenever you're in the lab if you want help / company :P.

arevans3, 2015-05-25:
You're probably going to have to do this stuff
https://wiki.archlinux.org/index.php/Remastering_the_Install_ISO
before you try to install arch. I can't find a full install anywhere.

Doomfest, 2015-05-25:
This is the net install one.
If you have the board connected through ethernet, you can pretty much skip the networking part if it autoconfigured properly.

https://wiki.archlinux.org/index.php/Beginners%27_guide

psyomn, 2015-05-25:
>
> If you have the board connected through ethernet,
Which is something we don't 100% have, and maybe tethering through usb
might not even be possible with an install (maybe it is? We don't know
yet). I'll be going over this tonight and see what exactly needs preparing.

psyomn, 2015-05-26:
Arielle and I took a look at stuff. We chose [zenwalk](http://en.wikipedia.org/wiki/Zenwalk). Looks good so far, and I've been reading some of the docs, which make me think that it was particularly made for embedded systems. I guess if it makes our lives miserable, we could switch again. Hopefully install will be done by tonight.

Doomfest, 2015-05-26:
What's wrong with arch?

psyomn, 2015-05-26:
Nothing! But we looked at some alternatives and zenwalk looks nice. You should've been here :P.

psyomn, 2015-05-26:
Zenwalk is installed, and looks quite good. I'm closing this - we have an OS and looks usable. We just need to setup internets again. We might get access to a router (MAYBE), but for now we're stuck with tethering. 

This satisfies everything from #2 (git, python ready to go), except internets. If we can repeat what was done last time, that would be good!

cc: @arevans3 @msnidal 

I'll be in lab tomorrow lateish if anyone is nearby.

closed 2015-05-26 by psyomn