space-concordia-robotics/robotics-rover #14: Add wrong value detection for movement in ucontroller manager before sending the commands.

Doomfest, 2015-06-18:
Need to detech if values received to the ucontr manager are valid before sending them to the ucontroller for movement.

Doomfest, 2015-06-20:
I added code which validates the values before sending them to the micro. @psyomn i was testing your client and the server was receiving everything . We need to check which numbers the client sends to the server. It seems like it sends all numbers and doesnt care about forward, reverse, turn left , right and stop movements.

psyomn, 2015-06-25:
I'm not sure I understand what you mean. The server understands what command is sent - that's why there are the constants, then the commands (patterns) which are executed, and finally fire the hooks if any exist.

Doomfest, 2015-07-24:
The validation code  happens within the ucontroller manager.

closed 2015-07-24 by Doomfest