space-concordia-robotics/robotics-rover #23: Set up ssh on the rover

msnidal, 2015-07-30:


psyomn, 2015-07-30:
Add tmux to the list as well!

msnidal, 2015-07-31:
openssh is installed and ssh is working over Sputnik. I also installed tmux.

closed 2015-07-31 by msnidal

psyomn, 2015-07-31:
Much appreciated, thanks.

Processes that we want to execute in the background may be started inside a
tmux session. I was able to do something similar with the old linux
"screen" command, so I'm very sure there is something available for tmux.

A quick search in google gives the following:

http://stackoverflow.com/questions/5609192/how-to-set-up-tmux-so-that-it-starts-up-with-specified-windows-opened

Take a look at the two first answers; the second one actually has the
commands we can reuse in a bash script to simply execute whenever the rover
is to be tested / deployed.