space-concordia-robotics/robotics-rover #25: Fix command delay lag.

Doomfest, 2015-08-06:
The mini rover seems to go crazy when it loses a packet or its delayed. 
The rover will have a driving state and video state. When the packet is received, the rover will go to its corresponding state. When the packet is delayed or lost, a timer in its own thread, will timeout and set the rover state to stop state. I decided on a timeout of 20ms which gives us 50 packets per second. We might need to increase or decrease this value depending on the latency of packets through the routers but not too slow since we want it to be responsive.

[command_delay_fix.pdf](https://github.com/space-concordia-robotics/robotics-rover/files/1209/command_delay_fix.pdf)

psyomn, 2015-08-06:
Before you do anything, I think Mark was going to tweak my crappy implementation to use the multiprocessing library correctly.

Doomfest, 2015-08-06:
We still need a timer though. It will make it more robust.

msnidal, 2016-01-23:
See #52 on robotics-networking - this is a networking issue, not on the rover end.

closed 2016-01-23 by msnidal