space-concordia-robotics/robotics-rover #28: Create rover state diagrams

msnidal, 2015-09-19:
We need to have a very good understanding of the rover states before we get further into the rover-side software development. The best way to do this is by creating a comprehensive state diagram for the rover that we can then implement.

psyomn, 2015-09-19:
This is also where self healing stuff may be added (as we talked about).

I think a good introduction is this paper maybe. There's another student
that might be interested on this topic.

http://www.cs.cmu.edu/afs/cs.cmu.edu/project/able/ftp/WOSS02/paper-submitted.pdf

(Note: If you're on campus, you can view similar papers via
scholar.google.com).

msnidal, 2015-09-30:
Do you want to work on this tomorrow @Doomfest?

Doomfest, 2015-10-24:
Done.

closed 2015-10-24 by Doomfest